ChibiOS/ChibiOS-Contrib #79: [onewire] Strange uint_fast8_t issue

russss, 2016-05-29:
I'm using the onewire driver on an STM32F042F6P6 with ARM-GCC 4.9.3. I was having issues with onewireReset always returning false. 

The [initial check](https://github.com/ChibiOS/ChibiOS-Contrib/blob/master/os/hal/src/hal_onewire.c#L650) that the bus was high was always failing, even though it was definitely high (verified with multimeter) and there was no bus activity. If I tried to check the state myself with `palReadPad`, it would always return 1.

It appears that is somehow related to the `uint_fast8_t` return type on `ow_read_bit`. Replacing this line:

```if (PAL_LOW == ow_read_bit(owp))```

With this:

```if ((uint_fast8_t)PAL_LOW == ow_read_bit(owp))```

Seems to fix the issue (although it still happens in numerous other places in the code). I have no idea why.

barthess, 2016-05-30:
Strange behaviour. This "uint_fast8_t" looks strange and unusual. I chose it because written 1-wire driver with 8-bit MCUs in mind but all users I know use it on STM32. Now I think to change it to "unsigned int".

barthess, 2016-05-30:
I updated driver. Could you test it?

russss, 2016-05-31:
Unfortunately even after fixing this particular issue there's something else wrong, and I'm probably going to need to get hold of a logic analyzer to look into it further.

closed 2022-02-07 by fpoussin