ChibiOS/ChibiOS-Contrib #84: qeiUpdate, unexepcted value due to counter wrapping

sdalu, 2016-06-25:
In the QEI driver, the qeiUpdate function returns the counter delta from the last reading, which when the counter is wrapping result in a huge delta.

| prev count | current count | delta | expected delta |
|---|---|----|----|
| 0 |65535 | 65535 | -1 |
| 65535|0|-65535|+1|

barthess, 2016-06-27:
I am not an author of QEI driver but I can review your pull request and merge it.

sdalu, 2016-06-28:
The fix looks like reverting commit https://github.com/ChibiOS/ChibiOS-Contrib/commit/4a36fb944b854eb91e74ce18a7b2cb0cedf7d0d1 . Initial code was relying on the wraping of the int16_t when substracting two uint16_t to provide the correct value. Which still open a can of worm:
- overflow of signed int is undefined behaviour in C/C++
- writting new driver impose to chose identical size for qeidelta_t and qeicnt_t and no mention of it is made
- value will still be wrong if delta is really above 32767 (int16_t)

I think qeiUpdate methods should be removed:
- values returned by qeiUpdate can be meaningless if counter has wrapped several times between requests
- delta values would be better provided by a driver callback when updating the counter, which will also make size of qeidelta_t and qeicnt_t independant. (possible with nRF51, but I don't know enough about STM32)
- if driver is not able to provide such a callback, user can still take the risk to implement its own version of current_value - last_value

fpoussin, 2017-01-04:
I merged a pull request that reverts the said commit. (among other things)

closed 2019-10-23 by fpoussin