ChibiOS/ChibiOS-Contrib #111: ChibiOS RT needed

romainreignier, 2017-02-02:
Hello,
I am using the HAL of ChibiOS but not the RT part because I do not need a RTOS for my projet.
I am using your QEI driver. It works nice but I had to move some:

```C
#include "ch.h"
```

after the line checking if the module is used. Ex:

```C
#if (HAL_USE_TIMCAP == TRUE) || defined(__DOXYGEN__)
```

The files I had to modify:

- `os/hal/include/hal_eeprom.h`
- `os/hal/include/hal_timcap.h`
- `os/hal/ports/STM32/LLD/DMA2Dv1/hal_stm32_dma2d.c`
- `os/hal/ports/STM32/LLD/LTDCv1/hal_stm32_ltdc.c`
- `os/hal/ports/STM32/LLD/TIMv1/hal_timcap_lld.c`

So if the RT part is really needed for these components, maybe it will be possible to move the `#include` after the check is the module is used in order to allow users to use other modules without the RT part of ChibiOS.

Thanks.

barthess, 2017-02-04:
I do not use that drivers and do not know how important RT for them (suppose not important) but I am not sure. Sad but contributors of that drivers looks vanished. We can just move #include "ch.h" into #if statement as you propose. This is safer way. Or we can try to delete this include statement at all and than check buildability of the affected application in testhal directory. This is more proper way.

fpoussin, 2017-02-06:
Hey,
I'll have a look at the timcap and eeprom drivers which I contributed a while ago.

fpoussin, 2017-02-06:
Fixed timcap and eeprom in commit 11e949d.

romainreignier, 2017-02-06:
@fpoussin Thanks!

commit 15517ffb (2017-03-01): [DMA2D, LTDC] Removing ch.h dependencies. Fix #111.
closed 2017-03-01 by barthess