ChibiOS/ChibiOS-Contrib #121: USB Host Driver Status / Progress

apmorton, 2017-07-04:
Hi All,

First thing, I want to thank @dismirlian for his work on the USB Host driver thus far - it is really a great improvement over other options in this space for STM32, and I know it is no small task to write such a complex piece of code.

I have been experimenting with the latest revisions of the driver on here and it looks promising - I am hoping to use it as a replacement to the HCC USB Host stack in an ongoing project where we are encountering issues due to some design flaws in HCC's stack.

Couple things that I have encountered so far:
- no support for out-of-tree drivers
  - need to modify the chibios-contrib code base itself to add any project specific drivers that may not be suitable for inclusion in chibios-contrib
- PTXFE not implemented, meaning ISO/INT OUT transfers result in IRQ storm and lock up system
  - copying implementation from NPTXFE seems to work for at least INT OUT transfers so far - once I confirm ISO works I will be happy to submit a PR

Diego, have you given any thought to potential API's for better control over the enumeration process without modifying the chibios-contrib code base itself?  I have hacked together some hooks to get up and running for now, but it is by no means pretty.  I can try to clean up what I have so far and publish it for review if that helps.

I am idling on #chibios @ freenode if anyone wants to discus these topics.

dismirlian, 2017-07-04:
Hi Austin,

Thanks for your feedback! I really appreciate the input from someone who really uses the driver.

You make a good point regarding out-of-tree class drivers. I had never thought of this use case, and it will certainly add flexibility to the driver. I had considered a dynamic registration scheme for class drivers, but I ruled it out because it goes against the design philosophy of ChibiOS (where everything is as "static" as it gets). However, I agree that we could add some functionality to the enumeration process to support out-of-tree class drivers. I'd be glad to see what you've come up with, and integrate it into the main code base.

Regarding ISO/INT out transfers, I didn't implement them because I don't have any hardware available to test, but, as you have noticed, the implementation should be very similar to NPTXFE. Of course, if your modifications work fine, we should also integrate them into the main code base.

Regards,
Diego.

dismirlian, 2017-07-09:
Hi,

It seems that the support for out-of-tree drivers was really a major issue... Someone else contacted me with the same request.

Please check https://github.com/dismirlian/ChibiOS-Contrib/tree/usbh_devel.

I've commited some changes to allow for out-of-tree class drivers. The code may not be definitive; input and suggestions are welcome. The changes are:

- New #define HAL_USBH_USE_ADDITIONAL_CLASS_DRIVERS in halconf_community.h (set to TRUE to enable out-of-tree class drivers).

- If you set this to TRUE, you must supply a usbh_additional_class_drivers.h file, which declares the additional class drivers (please check the STM32F4Discovery example in testhal/STM32/STM32F4xx/USB_HOST, namely the files usbh_custom_class_example.c/.h).

- In the STM32F4Discovery testhal project there is an example/skeleton class driver, which hints how you should integrate the driver into the USB Host framework (basically it's the same as the in-tree drivers).

The main change is that now the initialization function is part of the virtual methods table of the class driver (as should have from the beginning), so the main USB host driver can loop over the usbh_classdrivers_lookup table and initialize all the class drivers, including the out-of-tree ones.

Please check the example and tell me what you think, since your feedback is important to implement this feature in a way that is easy to use for the class driver developer.
 
Regards,
Diego.

closed 2019-10-23 by fpoussin