ChibiOS/ChibiOS-Contrib #124: mx2board.py issue with STM32F7XX .ioc file

m1keph0to, 2017-09-06:
Thanks for the tool, I picked it up as a part of the ChibiOS distribution then found the original on GitHub.

In the case of using STM32F7XX.ioc file 

`$ python mx2board2.py -m /c/Program\ Files/STMicroelectronics/STM32Cube  -p stm32f7_2.ioc -o board_gpio.h
Opening stm32f7_2.ioc
MCU is STM32F765N(G-I)Hx
Opened C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\db\mcu\
Opening GPIO file C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\db\mcu\\IP\GPIO-STM32F76x_gpio_v1_0_Modes.xml
Opening project file stm32f7_2.ioc
Traceback (most recent call last):
  File "mx2board2.py", line 332, in <module>
    defines = gen_defines(proj)
  File "mx2board2.py", line 273, in gen_defines
    ch_num = int(signal[-1:])
ValueError: invalid literal for int() with base 10: 'R'
`

The error is caused by the fact that the .ioc file contains  lines like 

`
SH.S_TIM2_CH1_ETR.0=TIM2_CH1,Input_Capture1_from_TI1
`

The regular expression at line 271 matches the pattern terminating _ETR rather than a digit. 
`
          defines['PORT_'+label] = 'GPIO' + port_key
            defines['PAD_'+label] = pad_key

            if re.search(r"TIM\d_CH\d**$**", signal, re.M):  <---- 
                timer = signal.replace('S_TIM', '').replace('_CH', '')[:-1]
                ch_num = int(signal[-1:])
`
There needs to be a '$' to only match complete strings at the end of the regular expression, this avoids matching names like TIM2_CH1_ETR. 
I am not sure whatever other sideffects there are of this change, it produced a valid-appearing board.h after changing this.

fpoussin, 2017-09-06:
Actually it should even be  r"TIM\d+_CH\d$" because we have more than 9 timers, thanks for catching this!

fpoussin, 2017-09-06:
Pushed a fix.

closed 2017-09-06 by fpoussin