ChibiOS/ChibiOS-Contrib #127: Tools running on the host system

funnydog, 2017-09-26:
Hello,

I'm porting ChibiOS to the Teensy 3.5 board and while it's nowhere near production quality, I'm at a point that the GPIOs and all the 4 UARTs work correctly. Also all the standard tests run successfully.

However I'm stuck with a problem relevant to the PLL configuration: every information I need to generate the PLL divisor and multiplier is known at compile time, yet I'm still computing their value at runtime on the MCU, similar to the way it's done in the K20x ports.

I could simplify everything by running a custom program on the host system that includes the needed headers and generates a header with the PLL divisor and multiplier. The latter would then be included directly by hal_lld.c which would correctly set the PLL. In the case no suitable combination of divisor and multiplier is found the tool would return an error and end the compilation.

My problem is I don't know the ChibiOS makefiles very well, so I added custom targets in my board.mk. Unfortunately I had to hardcode the gcc compiler, options, include paths and everything. I had to provide a fake first: all target to keep the existing make behaviour.

Is there a better way to do it? Should I stick to the way explained above?
Or should I compute the values for the PLL at runtime?

Thank you.

barthess, 2017-09-27:
It's better to ask Giovanni on official forum. As I know he plans to introduce some kind of "clock driver".

funnydog, 2017-09-29:
ok thank you

closed 2017-09-29 by funnydog