ChibiOS/ChibiOS-Contrib #134: [Question] USBH driver with external USB host controller

dogtopus, 2017-11-23:
I am planning to implement a driver of MAX3421E for ChibiOS. After doing some research, I think the best way to do it is to implement a complex driver that sits on top of the SPI driver. I would try to avoid writing a LLD driver for it since the LLD looks more like a place for "MCU-specific" code and the MAX3421E does not depend on the MCU. However, I also found that the USBH driver calls `usbh_lld_*` function of specific LLD implementation of a certain MCU, which means the external controller and the MCU built-in controller cannot possibly co-exist if I implement the `usbh_lld_*` in my driver (and also the code may look weird since `usbh_lld_*` appears in a complex driver).

So my question: Whatt is the best way to implement such driver that cooperates with USBH, or is it just a bad idea for the current USBH implementation?

barthess, 2017-11-23:
You can take a look at MMC over SPI and check how it coexists with SDIO driver. I'm not sure about compatibility of current USBH and MAX3421. May be it will require additional abstraction layer with reworking of USBH.

dismirlian, 2017-11-24:
You are right that the external controller and the MCU's OTG core won't be able to coexist without reworking the USBH driver. I think the steps towards the implementation would be:
- Avoid direct calls of the `usbh_lld*` functions, and replace them with calls through a VMT.
- Separate the `*_ll_data` structure members (see hal_usbh_lld.h) from the high-level USBH driver structures (for instance, `_usbh_ep_ll_data` extends `struct usbh_ep`). I think this is the most difficult step.

Regards,
D.

dogtopus, 2017-12-07:
@dismirlian Do you have any plan to do the split? I would rather not to do that for now since I don't have a STM32 board for test and do it blindly may cause serious regressions (especially since I am not 100% familiar with the USBH stack and STM32)

dismirlian, 2017-12-11:
Hi @dogtopus, currently I don't have much time to work on this. Probably, I'll be busy until February to do such "large" modifications to the driver. I'm sorry....

However, I *could* test your modifications on a STM32F4Discovery board.

dogtopus, 2017-12-21:
After reading through the USBH driver and other ChibiOS/HAL drivers that fits a similar model as the internal/external USBH controller (e.g. sdc and mmc_spi), I have 2 ideas on how to implement the split:

1. Create an interface for USBH drivers, move common code (e.g. device enumeration, descriptor parser, class driver loader, etc.) to a new file and write different drivers for internal and external USBH controller. IMO this is the more elegant way. However it involves creating and changing multiple files, introducing more data structures and may require rewriting some large functions as macros.
2. Keep the current USBH high level code, decouple the llds by constructing "subclasses of USBH" for them (i.e. vmt, sub data structure, similar to @dismirlian 's approach). Write object initializer and other necessary hardware-specific methods for the decoupled llds and use the USBH API on the "subclasses" (by casting the driver object to a generic type or use a void pointer to hold subclass-specific data and let the functions in vmt cast the pointer to a readable object) to complete the high level operation. This is simpler to implement but looks a bit hacky IMO.

What is your opinion on these? Or do you have any better ideas on the implementation?

Also some macros such as the `USBH_DECLARE_STRUCT_MEMBER` will be harder to implement after the decoupling due to different memory layout requirements on multiple drivers (internal controller may require certain buffers to align to specific boundary and these buffers might be present in the API instead of the lld). Any ideas on how to resolve such conflict?

closed 2021-08-19 by dogtopus