ChibiOS/ChibiOS-Contrib #155: Add support for ChibiOS 18.2

elagil, 2018-04-09:
Hello!

I am porting my application from Chibios 17 to 18 and I use the FSMC driver.

Problem:
The new build system catches ALLSRC and ALLINC paths for compilation. The "platform.mk" file of the contrib branch first includes the original "platform.mk" and then adds its own sources to

PLATFORMSRC
PLATFORMINC

However, these are not used by the makefile anymore. 

In my instance, I could fix it by changing the names like this:
```
include ${CHIBIOS}/os/hal/ports/STM32/STM32F7xx/platform.mk

ALLSCR += ${CHIBIOS_CONTRIB}/os/hal/ports/STM32/LLD/FSMCv1/hal_fsmc.c \
               ${CHIBIOS_CONTRIB}/os/hal/ports/STM32/LLD/FSMCv1/hal_nand_lld.c \
               ${CHIBIOS_CONTRIB}/os/hal/ports/STM32/LLD/FSMCv1/hal_fsmc_sram.c \
               ${CHIBIOS_CONTRIB}/os/hal/src/hal_fsmc_sdram.c

ALLINC += ${CHIBIOS_CONTRIB}/os/hal/ports/STM32/LLD/FSMCv1 \
               ${CHIBIOS_CONTRIB}/os/hal/ports/STM32/LLD
```

but this will break compilation with ChibiOS 17. 

Any comments on what would be the best solution?

Adrian

elagil, 2018-04-10:
I see that this issue was already raised here #151. Sorry for this duplicate.

fpoussin, 2018-05-17:
Closing this one

closed 2018-05-17 by fpoussin