ChibiOS/ChibiOS-Contrib #162: QEI driver warnings

vrepetenko, 2018-06-18:
Please fix:
ChibiOS version - 17.6.3
community/os/hal/include/hal_qei.h
```
 */
 typedef enum {
   QEI_OVERFLOW_WRAP    = 0,     /**< Counter value will wrap around.        */
-#if QEI_USE_OVERFLOW_DISCARD == TRUE
+#if defined(QEI_USE_OVERFLOW_DISCARD) && QEI_USE_OVERFLOW_DISCARD == TRUE
   QEI_OVERFLOW_DISCARD = 1,     /**< Counter doesn't change.                */
 #endif
-#if QEI_USE_OVERFLOW_MINMAX == TRUE
+#if defined(QEI_USE_OVERFLOW_MINMAX) && QEI_USE_OVERFLOW_MINMAX == TRUE
   QEI_OVERFLOW_MINMAX  = 2,     /**< Counter will be updated upto min or max.*/
 #endif
 } qeioverflow_t;
```

fpoussin, 2018-06-19:
Can you submit a pull request since you already made the changes?
Thanks

vrepetenko, 2018-06-19:
Ok! I will submit it.

closed 2018-06-23 by fpoussin