ChibiOS/ChibiOS-Contrib #188: USB Host fixes problem

dismirlian, 2019-03-11:
Hi all,

I think a bug was introduced in commit https://github.com/ChibiOS/ChibiOS-Contrib/commit/f014b8da68f162bb06ca88acf0085382a76d8dcd. I'm not 100% sure, and would appreciate some input from the author @josesimoes. Originally, `_scsi_perform_transaction` returned `MSD_RESULT_FAILED` when `transaction->csw_status` was `CSW_STATUS_FAILED`, after performing a SCSI Request Sense. The commit introduced line 520, which would cause `_scsi_perform_transaction` to return `MSD_RESULT_OK` if the request sense was successful, even if the transaction was not carried out. I think the original code is correct. José, any comments?

https://github.com/ChibiOS/ChibiOS-Contrib/blob/a4388d61308f64b7adffeb62417fc0cb2e739a3f/os/hal/src/usbh/hal_usbh_msd.c#L502-L527

Thanks,
Diego.

josesimoes, 2019-03-11:
Hi @dismirlian ,

I've added that fix to overcome an intermittent issue I was having when the transaction was failing _AND_ the subsequent scsi_requestsense was true. 
With the original code, on transaction FAILED and following requestsense OK, the return always ended up being MSD_RESULT_FAILED which I don't think is correct.

As it is now, I stopped experiencing that issue. Maybe my fix wasn't the best...

fpoussin, 2019-03-11:
Could you build a test case for this to verify if it's an actual bug or more of a workaround? 
Also, it's always good to add comments in the code to get some context.

josesimoes, 2019-03-11:
Where should I add that test case?

Also I have to make a forward statement saying that I'm not an expert on what concerns USB low level protocol so I would rather have someone with knowhow on this to jump in the conversation...

dismirlian, 2019-03-11:
Hi @josesimoes, actually I wrote this code ~4 years ago, and I don't remember the details, but:
- CSW_STATUS_FAILED indicates that the operation has failed, so the Host should do some recovery.
- All of the possible failure cases are covered in https://usb.org/sites/default/files/usbmassbulk_10.pdf, section 6.5 ("Host Error Handling") through 6.7 ("The Thirteen Cases").
- I decided to do a very simple proof-of-concept error handling (auto-sense only), and inform the upper layer of the operation failure.

So, I don't think that we should inform the upper layer that everything is OK, when in fact something has failed. IMHO a successful REQUEST SENSE does not indicate that the previous operation has succeeded.

The "clean" solution would be to implement the error recovery as per the USB spec.

Are you seeing any write or read errors?

Regards,
D.

fpoussin, 2019-03-11:
This indeed should follow the USB MSD spec as close as possible instead of hiding errors from the applications.
Failure when talking to peripherals is normal and should be taken care of.

@josesimoes anywhere in the testhal/demo code is fine as long as it reproduces the error.

josesimoes, 2019-03-11:
@dismirilian and you did a great job, btw! 

The problem was in the initial steps of the driver state machine, before the mount. I can't say exactly what was the root cause because it was happening randomly.

If you guys are not comfortable with this, by all means, do revert the change!

dismirlian, 2019-03-12:
Hi @josesimoes, I'll see if I can implement the "thirteen cases", so we can have proper error handling. I think the original code is too "pessimistic".

D.

fpoussin, 2019-03-26:
@dismirlian did you have any time to look into it?

fpoussin, 2021-02-14:
closing this as it's not present in the code anymore

closed 2021-02-14 by fpoussin