ChibiOS/ChibiOS-Contrib #202: COMPv1 and stm32L031

NickNifontov, 2019-10-23:
Hi, i need a help of ChibiOS's Contrib community.

I am newbie in ChibiOS architectory. So faced with promblem when adopting COMPv1 to stm32L031 (M0).

In stm32l031 (Mo) Comparator uses interrupt - ADC1_COMP_IRQHandler same that a ADC.

But in COMPv1 - STM32_COMP_1_2_3_IRQ_PRIORITY (COMP_1) and and STM32_COMP_1_2_3_IRQ_PRIORITY (COMP_2).

As i understand ChibiOS HAL use ADC1_COMP_IRQHandler for ADC HAL Driver? Or i can change library to use ADC1_COMP_IRQHandler?

fpoussin, 2019-10-23:
Hi,
STM32L0s are no supported by the current driver, I'll make some changes when I have time.

NickNifontov, 2019-10-23:
My regards. It will be awesome.

I can perform all tests at stm32l031f6
Thanks

fpoussin, 2019-10-29:
I've updated the driver, give it a try.

fpoussin, 2019-10-31:
@NickNifontov any update?

NickNifontov, 2019-10-31:
I'm on a business trip now. On Saturday I will check everything. thank

fpoussin, 2019-11-04:
@NickNifontov did you get to try?

NickNifontov, 2019-11-04:
all is Work !!!
thanks

closed 2019-11-04 by fpoussin