ChibiOS/ChibiOS-Contrib #210: SDRAM start leads to panic (SV4)

elagil, 2020-01-06:
The current version of SDRAM drivers breaks execution when calling `sdramStart()` because of nested calls to `osalSysLock()`. After `sdram_lld_start() -> sdram_lld_init_sequence()`, the thread is put to sleep with `osalThreadSleepMilliseconds()`, which also contains a `osalSysLock()` statement, leading to the SV4 panic.

I suggest to remove the lock statements from `sdramStart()` to restore functionality. Are they actually required? If not, I can create a pull request.

fpoussin, 2020-01-06:
Hi,

Thanks for finding this bug.
osalThreadSleepMilliseconds could be replaced by **osalSysPolledDelayX**.

The start/stop function locks are needed for driver state consistency.

elagil, 2020-01-07:
Good to know, I now replaced the delay with and do some testing.

`osalSysPolledDelayX(TIME_US2I(100))`

fpoussin, 2020-01-07:
Feel free to open a pull request once testing is done

closed 2020-01-09 by fpoussin