ChibiOS/ChibiOS-Contrib #224: I get chSysHalt("SV#6") while running ChibiOS-Contrib-RT-STM32F407-DISCOVERY-RTT+SystemView Example

abusous2000, 2020-10-02:
Hi guys

I decided to evaluate **Segger J-Link,** and tested the example in the [ChibiOS-Contrib section](https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x/demos/STM32/RT-STM32F407-DISCOVERY-RTT%2BSystemView)

However, when I run it in the debugger; it is as if ChibiOS thinks it is running in a **thread contexts** it freezes at chSysHalt("SV#6")

Clearly I can see in the debugger that I am in a DMA ISR , which invokes macro **CH_CFG_IRQ_PROLOGUE_HOOK()** that is delegated to **SEGGER_SYSVIEW_RecordEnterISR()** method 
Here is a picture
[image]

Therefore, I wonder if this is a bug, or my J-Link hardware is bad, or I did something wrong?

Many thanks in advance.

abusous2000, 2020-10-03:
BTW,

Once I changed these macros in file **SEGGER_SYSVIEW_ChibiOS.h** to the following, it did work well with no exceptions. You will find in bold what I have changed.
`#define CH_CFG_IRQ_PROLOGUE_HOOK() {                     \
  **_dbg_check_enter_isr();**                                 \
  SEGGER_SYSVIEW_RecordEnterISR();                        \
  _CH_CFG_IRQ_PROLOGUE_HOOK();                            \
}

#define CH_CFG_IRQ_EPILOGUE_HOOK() {                      \
  **_dbg_check_leave_isr();**                           \
  _CH_CFG_IRQ_EPILOGUE_HOOK();                            \
  port_lock_from_isr();                                   \
  _dbg_enter_lock();                                      \
  if (_isr_is_tail() && chSchIsPreemptionRequired()) {    \
    SEGGER_SYSVIEW_RecordExitISRToScheduler();            \
  } else {                                                \
    SEGGER_SYSVIEW_RecordExitISR();                       \
  }                                                       \
  _dbg_leave_lock();                                      \
 }`

the sad thing that SystemView now reports: **Could not find RTT Control Block.Timeout: 20000 ms** which isn't your problem.
thx

troyerta, 2021-01-15:
I can confirm this workaround seems to fix the issue from the state checker's perspective. The JLinkRTTClient still works fine, but SystemView reports "ERROR: Could not find SystemView Buffer". No idea why this context check would corrupt it, but I'll try pointing to it directly to see how SystemView responds.

fpoussin, 2021-02-14:
Closing this as the fix is merged

closed 2021-02-14 by fpoussin