ChibiOS/ChibiOS-Contrib #225: Add SN32F248B Support

SmileYzn, 2020-10-24:


Codetector1374, 2021-01-05:
Hi, do you have a dev board / or some way that I can obtain a chip so I can do the port? I don't really want to port without a good testing platform.

SmileYzn, 2021-01-07:
I have a Motospeed CK62 keyboard, i did not know if can use it

Codetector1374, 2021-01-08:
I can't see where I can source the Chip directly (So I can design a prototype dev board or something). But I don't have such a keyboard that uses this chip. Although inconvenient a keyboard is a valid board to test ChibiOS port on. I can help with a port if I can get my hand on something that uses this chip. (Or where to source such a chip)

dexter93, 2021-11-17:
Hi, sorry for the bump. We use these chips on various chinese keyboards. Trying to get QMK working on it through ChibiOS

A list with some known keyboards using these chips can be found [here](https://github.com/SonixQMK/Mechanical-Keyboard-Database)

Alternatively, a dev board can be found [here](https://gwell.en.taiwantrade.com/product/starter-kit-32f248b-1982342.html)

While you can track our progress in ChibiOS porting [here](https://github.com/SonixQMK/ChibiOS-Contrib)

closed 2022-04-18 by fpoussin