ChibiOS/ChibiOS-Contrib #252: Question: How should peripheral symbols numbered?

elfmimi, 2021-01-04:
Related to https://github.com/ChibiOS/ChibiOS-Contrib/pull/235.
I would like these numbered in straight forward way. What does the design guide say?
@fpoussin @alexclewontin
```
NUC123_I2C1_GLOBAL_HANDLER
NUC123_I2C1_GLOBAL_NUMBER

NUC123_I2C2_GLOBAL_HANDLER
NUC123_I2C2_GLOBAL_NUMBER

NUC123_HAS_I2C0
NUC123_HAS_I2C1

NUC123_I2C_USE_I2C0
NUC123_I2C_USE_I2C1

I2CD1
I2CD2
```

alexclewontin, 2021-01-04:
I have no idea what the official guidance on this is, but in the serial driver I mapped SD0 -> UART0 and SD1 -> UART1, as mapping SD1 -> UART0 and SD2 -> UART1 seemed way too confusing. To try to respect the convention of PERIPHERAL_DRIVER1 being the "main" or "default" driver object, I set SD1 as active by default, and SD0 as inactive by default, although this can of course be configured using defines.

If you decide to do differently, I can change the serial driver to match whatever you do.

I don't know what _GLOBAL_ is supposed to signify. It seems wrong, and so if it were up to me, I'd just change it to `NUC123_I2Cn_HANDLER` and `NUC123_I2Cn_NUMBER`

Codetector1374, 2021-01-05:
I also used 1 for my LPC ports. Since I see the most of the offically supported STM32 drivers all seems to start from 1?

elfmimi, 2021-01-05:
Thanks you two for your response.

Let me see... , as of LPC, UART and many of peripherals are unnumberd. and you alligned number 1 for them. for ex. UARTD1. maybe this is okay.
but it looks you assigned SPID1 to LPC_SSP1. and support for LPC_SSP0 is yet to come.
what is your plan for LPC_SSP0 ? SPID0 or SPID2 ?

Well, I made up my mind to use straight forward I2Cn to I2CDn mapping.
I believe users of ChibiOS know better than to just write raw symbol names everywhere but to use pointers.
I believe it's better to avoid confusing configuration at any cost or you will definitely have payback later.

elfmimi, 2021-01-05:
I'm closing this.

closed 2021-01-05 by elfmimi