ChibiOS/ChibiOS-Contrib #254: [Feature Request] Support for Raspberry Pi Pico

ryanm101, 2021-01-21:
https://www.raspberrypi.org/blog/raspberry-pi-silicon-pico-now-on-sale/

Support here for this is a pre-req for QMK (https://github.com/qmk/qmk_firmware/issues/11649)

stapelberg, 2021-04-20:
Just in case anyone finds this issue and not the other forum thread: support for the Raspberry Pi Pico seems to be added to ChibiOS itself (not ChibiOS-Contrib). See https://forum.chibios.org/viewtopic.php?f=3&t=5800

robin-jones, 2021-06-28:
If someone can update the mirror maintenance script for stable_21.6.x, I believe this board is now supported... @fpoussin?

GrzegorzZajac000, 2021-07-31:
Any news on support for rpi Pico?

fpoussin, 2021-09-17:
https://github.com/ChibiOS/ChibiOS/tree/master/demos/RP/RT-RP2040-PICO
Closing this as it's in the work

closed 2021-09-17 by fpoussin

Corallus-Caninus, 2022-04-06:
is this still in the works or finished? I haven't seen activity on the linked repository.