ChibiOS/ChibiOS-Contrib #255: [Feature Request] Support for NUC121

jindaxia, 2021-01-25:
http://www.nuvoton.com.cn/products/microcontrollers/arm-cortex-m0-mcus/nuc121-125-series/

elfmimi, 2021-01-29:
NUC120 NUC121 NUC122 NUC123 NUC125 NUC126 and NUC140 NUC220 NUC240 and more... these are all close siblings with USB support. I'd like to see support for them all in the future.

elfmimi, 2021-02-05:
Here is my progress so far.
https://docs.google.com/spreadsheets/d/1X-sQdVe6JJeprrTbHXPAz0ckdHqr3YHeE0xE8iY3nFo
I suppose LLD of SPI for NUC121/NUC125/NUC126 is likely get named SPIv2 .

elfmimi, 2021-03-21:
Here's an experimental support for NUC125 and NUC121 https://github.com/elfmimi/ChibiOS-Contrib/tree/nuc125-hacky .
This is an attempt to find out how different those are from NUC123.

jazlee, 2021-06-26:
> Here's an experimental support for NUC125 and NUC121 https://github.com/elfmimi/ChibiOS-Contrib/tree/nuc125-hacky .
> This is an attempt to find out how different those are from NUC123.

any update on this?

elfmimi, 2021-06-26:
No update.