ChibiOS/ChibiOS-Contrib #307: Roadmap for ChibiOS-Contrib 21.11.x

dexter93, 2021-11-17:
ChibiOS has released a stable 21.11.x branch. Is there a roadmap on a -Contrib version?

fpoussin, 2021-11-27:
Hi,
I need to finish updating the repo for 21.6.x first, then I'll make the change directly to 21.11.x as it's the latest stable.

tzarc, 2021-11-28:
Fair enough; Giovanni has mentioned that he's "cancelled" the 21.6 stream due to too many issues.

tzarc, 2021-12-08:
Just an FYI from ChibiOS Discord:
[image]

zykrahgaming, 2021-12-31:
@fpoussin Hi, are you still working on this? If not, do you have a list of what needs to be done for 21.11.x to be completed?

zykrahgaming, 2021-12-31:
Also, i noticed that a lot of the changes regarding rp2040 were merged from @hanya. Though upon analysing the current state of the 21.6.x fork, a lot of the dependencies for the RP2040 are missing, as the rest of the MCUs in /testhal are still using ChibOS stable 20.3.x (as from the build.yml). hanya's fork looks like it uses the master branch of chibios though?

zykrahgaming, 2022-01-01:
https://github.com/ChibiOS/ChibiOS-Contrib/pull/310
I have made a PR which has updated ChibiOS-Contrib to the ChibiOS master branch (21.11.x)

closed 2022-02-06 by fpoussin