ChibiOS/ChibiOS-Contrib #313: So many failing branches?

rusefillc, 2022-02-01:
As of today:

https://github.com/ChibiOS/ChibiOS-Contrib fails tests, no changes for a few months
https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x fails tests, no changes for a few months
https://github.com/ChibiOS/ChibiOS-Contrib/commits/chibios-21.6.x fails tests, no changes for a few months

Is build system broken or is not just not important?

zykrahgaming, 2022-02-01:
It's falling due to a few reasons... See #310

rusefillc, 2022-02-01:
https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x is probably not caused by #310

zykrahgaming, 2022-02-01:
That PR just addresses the issues in both 20.3.x and 21.6.x that stop them from passing CI

rusefillc, 2022-02-01:
[image]

https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x failing state is probably not caused by #310?

fpoussin, 2022-02-06:
Failing branch can be caused by changed in the Chibios base code
As for 21.6.x, it's not been fixed, we'll move on to 21.11 directly.

rusefillc, 2022-02-06:
what has changed for https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x ? why is https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x no longer red?

fpoussin, 2022-02-06:
I've cleared some stall GH actions workflows

fpoussin, 2022-02-06:
all fixed now

closed 2022-02-06 by fpoussin