ChibiOS/ChibiOS-Contrib #327: Problem with Keychron K3V2 usb connection

seyahdoo, 2022-07-02:
I was having issues with SonixQMK on my Keychron K3V2, when i plug the keyboard to my machine it was sometimes not working completely and i had to replug it several times to get it to work.

While I was investigating this, I found that commit ff551265b12de68f2943b7d4c4793a33c8626d35 has broken this keyboards usb behaviour. It is consistently working correctly on d5e4fb2517b47a6a507fb765d3c78ea46b80f430 but it does not work on ff551265b12de68f2943b7d4c4793a33c8626d35.

Please do investigate further.

seyahdoo, 2022-07-10:
I tried doing this once again and I could not reproduce this behaviour. I belive it was caused by something else. It still occurs on latest SonixQMK sn32_master (a46e96cdd01cffc32d369c9317808799166f1ace) though.

I worked around this problem with adding "EEPROM_RESET" keycode to my keymap, If the keyboard does not work when I plug it in, I just press "EEPROM_RESET" and it connects after that.

seyahdoo, 2022-07-10:
And then I worked around having to press Fn+Space everytime I plug in with adding `eeconfig_init();` on `keyboard_post_init_user`. Now it works everytime I plug in the keyboard.

```
void keyboard_post_init_user(void) {
    eeconfig_init();
}
```

dexter93, 2022-11-25:
Hi there. Investigation is ongoing. It seems to be host USB controller related, depending on platform. Could you please report back on your system specs? And if it is still present with the latest commits? Thanks

seyahdoo, 2022-11-25:
Hello @dexter93, thanks for spending time on this issue. You can find my pc specs down below. If you need any additional information, please do ask here.

my specs are;
ASUS G14 GA401QM-HZ242T
AMD Ryzen 9 5900HS with Radeon Graphics
RTX 3060 (Laptop)

seyahdoo, 2022-11-25:
Also good to note:
https://seyahdoo.com/blog/qmk-on-keychron-k3v2/
https://github.com/seyahdoo/k3-v2-optical-qmk

seyahdoo, 2022-11-25:
Now that i have looked at the code again, my latest firmware Im using on my keyboard is using "#define NO_USB_STARTUP_CHECK" and not using "eeconfig_init();". I will recompile with latest sn32 develop now.

seyahdoo, 2022-11-25:
tried with latest sn32 develop, it does work with `#define NO_USB_STARTUP_CHECK` defined

seyahdoo, 2022-11-25:
tried with latest sn32 develop, and `#define NO_USB_STARTUP_CHECK` **not defined**, and it does not work 6/7 times when you plug in. When it didnt work on the first plug in, It did work after "EEPROM_RESET" keycode.

dexter93, 2022-11-25:
> tried with latest sn32 develop, and `#define NO_USB_STARTUP_CHECK` **not defined**, and it does not work 6/7 times when you plug in. When it didnt work on the first plug in, It did work after "EEPROM_RESET" keycode.

Interesting results. Can you please also expand this test by adding this https://github.com/SonixQMK/ChibiOS-Contrib/pull/48 ?

seyahdoo, 2022-11-25:
```
[seyyid@DESKTOP-UMH0T25 ~]$ qmk compile -kb keychron/k3/rgb/optical_iso -km iso_seyahdoo
Ψ Compiling keymap with make --jobs=1 keychron/k3/rgb/optical_iso:iso_seyahdoo

QMK Firmware 0.7.101
Making keychron/k3/rgb/optical_iso with keymap iso_seyahdoo

arm-none-eabi-gcc.exe (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/keychron/k3/../../../drivers/led/sn32/matrix_sn32f24xx.c                       [OK]
Compiling: keyboards/keychron/k3/k3.c                                                               [OK]
Compiling: keyboards/keychron/k3/rgb/optical_iso/optical_iso.c                                      [OK]
Compiling: keyboards/keychron/k3/keymaps/iso_seyahdoo/keymap.c                                      [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/send_string.c                                                                    [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_macro.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/bootmagic_lite.c                                                       [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: lib/printf/printf.c                                                                      [OK]
Compiling: drivers/eeprom/eeprom_driver.c                                                           [OK]
Compiling: platforms/chibios/eeprom_sn32.c                                                          [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgb_matrix/rgb_matrix.c                                                          [OK]
Compiling: quantum/rgb_matrix/rgb_matrix_drivers.c                                                  [OK]
Compiling: lib/lib8tion/lib8tion.c                                                                  [OK]
Compiling: drivers/led/sn32/rgb_matrix_sn32f24xb.c                                                  [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/dip_switch.c                                                                     [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt0_v6m.S                           [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/vectors.S                            [OK]
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v6m.S                        [OK]
Compiling: tmk_core/protocol/host.c                                                                 [OK]
Compiling: tmk_core/protocol/report.c                                                               [OK]
Compiling: tmk_core/protocol/usb_device_state.c                                                     [OK]
Compiling: tmk_core/protocol/usb_util.c                                                             [OK]
Compiling: platforms/chibios/platform.c                                                             [OK]
[... cut ...]
      |             ^~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:457: .build/obj_keychron_k3_rgb_optical_iso/lib/chibios-contrib/os/hal/ports/SN32/LLD/SN32F2xx/USB/hal_usb_lld.o] Error 1
Make finished with errors
make: *** [Makefile:478: keychron/k3/rgb/optical_iso:iso_seyahdoo] Error 1
```

trying only changing the /os/hal/ports/SN32/LLD/SN32F2xx/USB/hal_usb_lld.h file on sn32_master havent compiled. will try with that PRs commit now.

seyahdoo, 2022-11-25:
tried again in a different way and it did compile on sn32_master with only `/os/hal/ports/SN32/LLD/SN32F2xx/USB/hal_usb_lld.h` change now. But the issue persists, with this firmware it only worked `7/50` times on first plugging.

dexter93, 2022-11-25:
> tried again in a different way and it did compile on sn32_master with only `/os/hal/ports/SN32/LLD/SN32F2xx/USB/hal_usb_lld.h` change now. But the issue persists, with this firmware it only worked `7/50` times on first plugging.

This PR needs rebasing to target `sn32_develop` and should be tested against that branch. The issue is known to affect only cold boot/usb replug conditions

seyahdoo, 2022-11-25:
ok, will checkout to sn32_develop and try there

seyahdoo, 2022-11-25:
copied keychron folder from sn32_master
checkout to sn32_develop
deleted all keyboards
pasted keychron folder
overriten keychron folder with my config
didnt compile

```
[seyyid@DESKTOP-UMH0T25 ~]$ qmk compile -j=8 -kb keychron/k3/rgb/optical_iso -km iso_seyahdoo
Ψ Compiling keymap with make --jobs=8 --output-sync=target keychron/k3/rgb/optical_iso:iso_seyahdoo

QMK Firmware 0.7.101
Making keychron/k3/rgb/optical_iso with keymap iso_seyahdoo

 * No bootloader specified. Please set an appropriate bootloader in your rules.mk or info.json.        [ERRORS]

platforms/chibios/bootloader.mk:125: *** Invalid BOOTLOADER.  Stop.
Make finished with errors
make: *** [Makefile:414: keychron/k3/rgb/optical_iso:iso_seyahdoo] Error 1
```

seyahdoo, 2022-11-25:
there is no k3 folder on sn32_develop

dexter93, 2022-11-25:
> copied keychron folder from sn32_master checkout to sn32_develop deleted all keyboards pasted keychron folder overriten keychron folder with my config didnt compile
>
> ```
> [seyyid@DESKTOP-UMH0T25 ~]$ [...]

qmk-side `sn32_develop` is mid refactoring and currently has no sn32 keyboards inherited. things in there are bleeding edge and quite a few stuff are still missing ( including optical support). Please hop on the discord server, debugging this through github will take ages

seyahdoo, 2022-11-25:
ok, im at the sonix keyboard discord voice channel. we can debug there for an hour if you like.

dexter93, 2023-06-02:
small update here, this might just do it https://github.com/dexter93/ChibiOS-Contrib/commit/cbd841a3d02c37ece9777a90d055dee595c86643

seyahdoo, 2023-06-19:
im gonna try it soon, once i got home.

seyahdoo, 2023-07-25:
made k3 settings on sn32_develop to test the new code
https://github.com/SonixQMK/qmk_firmware/pull/341

seyahdoo, 2023-07-25:
Finally got time to test this, sorry for the delay. Here is my findings.

-------

Tested SonuixQMK/qmk_firmware/sn32_master(917a76c2b5c906f7f03ab07fc32399dc6ce9af15)
On AMD Ryzen 5 7600X Desktop -> Worked everytime. tried 5 times, No issues
On AMD Ryzen 9 5900HS Laptop -> tried to plug in 20+ times, no response. Keyboard lights up but pc wont respond.

-------

Tested SonuixQMK/qmk_firmware/sn32_master(917a76c2b5c906f7f03ab07fc32399dc6ce9af15) -> with my keyboard layout and #define NO_USB_STARTUP_CHECK = yes

On AMD Ryzen 5 7600X Desktop -> Worked everytime. tried 5 times, No issues.
On AMD Ryzen 9 5900HS Laptop -> Worked everytime. tried 5 times, No issues.

-------

Trying to Test SonuixQMK/qmk_firmware/sn32_develop(db0eb142ff897537e09611b09a68e01c62e7e189)
tried to copy keyboards/keychron/k3 from sn32_master to sn32_develop and compile, it wont compile.

```
[seyyid@Beast ~]$ qmk compile -kb keychron/k3/rgb/optical_iso -km iso_seyahdoo
Ψ Compiling keymap with make --jobs=1 keychron/k3/rgb/optical_iso:iso_seyahdoo

QMK Firmware 0.7.101
⚠ keychron/k3/rgb/optical_iso: DEVICE_VER in config.h is deprecated in favor of `usb.device_version` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: MANUFACTURER in config.h is deprecated in favor of `manufacturer` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: PRODUCT in config.h is deprecated in favor of `keyboard_name` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: PRODUCT_ID in config.h is deprecated in favor of `usb.pid` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: VENDOR_ID in config.h is deprecated in favor of `usb.vid` in info.json and will be removed at a later date
☒ keychron/k3/rgb/optical_iso: No LAYOUTs defined! Need at least one layout defined in info.json.
Making keychron/k3/rgb/optical_iso with keymap iso_seyahdoo

⚠ keychron/k3/rgb/optical_iso: DEVICE_VER in config.h is deprecated in favor of `usb.device_version` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: MANUFACTURER in config.h is deprecated in favor of `manufacturer` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: PRODUCT in config.h is deprecated in favor of `keyboard_name` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: PRODUCT_ID in config.h is deprecated in favor of `usb.pid` in info.json and will be removed at a later date
⚠ keychron/k3/rgb/optical_iso: VENDOR_ID in config.h is deprecated in favor of `usb.vid` in info.json and will be removed at a later date
☒ keychron/k3/rgb/optical_iso: No LAYOUTs defined! Need at least one layout defined in info.json.

 * No bootloader specified. Please set an appropriate bootloader in your rules.mk or info.json.        [ERRORS]

platforms/chibios/bootloader.mk:131: *** Invalid BOOTLOADER.  Stop.
Make finished with errors
make: *** [Makefile:392: keychron/k3/rgb/optical_iso:iso_seyahdoo] Error 1
```

-------

Resetted qmk_firmware folder to SonuixQMK/qmk_firmware/sn32_master(917a76c2b5c906f7f03ab07fc32399dc6ce9af15)
Deleted lib/ChibiOS-Contrib
Cloned dexter93/ChibiOS-Contrib/usb_quirks(cbd841a3d02c37ece9777a90d055dee595c86643)
qmk compile -kb keychron/k3/rgb/optical_iso -km iso

On AMD Ryzen 5 7600X Desktop -> Worked everytime. tried 5 times, No issues.
On AMD Ryzen 9 5900HS Laptop -> tried to plug in 20 times, no response. Keyboard lights up but pc wont respond.
 
-------

Resetted qmk_firmware folder to SonuixQMK/qmk_firmware/sn32_develop(db0eb142ff897537e09611b09a68e01c62e7e189)
copied k8 keyboard settings as k3
edited the keyboard config to make it work
qmk compile -kb keychron/k3/rgb/optical_iso -km iso

On AMD Ryzen 5 7600X Desktop -> Worked everytime. tried 5 times, No issues.
On AMD Ryzen 9 5900HS Laptop -> Worked everytime. tried 5 times, No issues.
 
-------

Summary:
sn32_develop works and does not need the new chibios contrib change to work.

dexter93, 2023-07-25:
issue seems to have been ironed out with #371 and  #376 . Some edge case scenarios do seem to persist, but it's qmk USB handling related - soon to be fixed