ChibiOS/ChibiOS-Contrib #378: LPC11uxx compile issue

crashniels, 2023-07-26:
Currently when compiling for qmk it will not compile unless you uncomment [this line](https://github.com/qmk/ChibiOS-Contrib/blob/553b773b151c2446bcf485aeb4c74cab7c082601/os/hal/ports/LPC/LLD/GPIO/hal_pal_lld.h#L48C15-L48C15). 

Is this a bug or am I missing something? I have mentioned it in https://github.com/qmk/qmk_firmware/pull/21621.

dexter93, 2023-07-29:
might be helpful to post the relevant build failure output log

crashniels, 2023-07-29:
```➜  qmk_firmware git:(k70_testing) ✗ qmk compile -kb corsair/k70_vengeance_rgb -km default
Ψ Compiling keymap with make --jobs=1 corsair/k70_vengeance_rgb:default

QMK Firmware 0.21.6
Making corsair/k70_vengeance_rgb with keymap default

Generating: .build/obj_corsair_k70_vengeance_rgb_default/src/info_deps.d                            [OK]
arm-none-eabi-gcc (Arch Repository) 13.1.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_corsair_k70_vengeance_rgb_default/src/config.h                               [OK]
Generating: .build/obj_corsair_k70_vengeance_rgb_default/src/keymap.c                               [OK]
Generating: .build/obj_corsair_k70_vengeance_rgb/src/info_config.h                                  [OK]
Generating: .build/obj_corsair_k70_vengeance_rgb/src/default_keyboard.c                             [OK]
Generating: .build/obj_corsair_k70_vengeance_rgb/src/default_keyboard.h                             [OK]
Compiling: keyboards/corsair/k70_vengeance_rgb/k70_vengeance_rgb.c                                  [OK]
Compiling: .build/obj_corsair_k70_vengeance_rgb/src/default_keyboard.c                              [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                        In file included from ./lib/chibios/os/hal/include/hal_pal.h:174,
                 from ./lib/chibios/os/hal/include/hal.h:306,
                 from platforms/chibios/gpio.h:18,
                 from platforms/gpio.h:21,
                 from quantum/matrix.h:22,
                 from quantum/matrix.c:21:
quantum/matrix.c: In function 'setPinOutput_writeLow':
platforms/chibios/gpio.h:28:57: error: 'PAL_MODE_OUTPUT_PUSHPULL' undeclared (first use in this function); did you mean 'PAL_MODE_OUTPUT_OPENDRAIN'?
   28 | #define setPinOutputPushPull(pin) palSetLineMode((pin), PAL_MODE_OUTPUT_PUSHPULL)
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
./lib/chibios-contrib/os/hal/ports/LPC/LLD/GPIO/hal_pal_lld.h:368:34: note: in definition of macro 'pal_lld_setpadmode'
  368 |   _pal_lld_setpadmode(port, pad, mode)
      |                                  ^~~~
./lib/chibios/os/hal/include/hal_pal.h:809:3: note: in expansion of macro 'palSetPadMode'
  809 |   palSetPadMode(PAL_PORT(line), PAL_PAD(line), mode)
      |   ^~~~~~~~~~~~~
platforms/chibios/gpio.h:28:35: note: in expansion of macro 'palSetLineMode'
   28 | #define setPinOutputPushPull(pin) palSetLineMode((pin), PAL_MODE_OUTPUT_PUSHPULL)
      |                                   ^~~~~~~~~~~~~~
platforms/chibios/gpio.h:30:27: note: in expansion of macro 'setPinOutputPushPull'
   30 | #define setPinOutput(pin) setPinOutputPushPull(pin)
      |                           ^~~~~~~~~~~~~~~~~~~~
quantum/matrix.c:80:9: note: in expansion of macro 'setPinOutput'
   80 |         setPinOutput(pin);
      |         ^~~~~~~~~~~~
platforms/chibios/gpio.h:28:57: note: each undeclared identifier is reported only once for each function it appears in
   28 | #define setPinOutputPushPull(pin) palSetLineMode((pin), PAL_MODE_OUTPUT_PUSHPULL)
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
./lib/chibios-contrib/os/hal/ports/LPC/LLD/GPIO/hal_pal_lld.h:368:34: note: in definition of macro 'pal_lld_setpadmode'
  368 |   _pal_lld_setpadmode(port, pad, mode)
      |                                  ^~~~
[... cut ...]
   30 | #define setPinOutput(pin) setPinOutputPushPull(pin)
      |                           ^~~~~~~~~~~~~~~~~~~~
quantum/matrix.c:87:9: note: in expansion of macro 'setPinOutput'
   87 |         setPinOutput(pin);
      |         ^~~~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:361: .build/obj_corsair_k70_vengeance_rgb_default/quantum/matrix.o] Error 1
Make finished with errors
make: *** [Makefile:392: corsair/k70_vengeance_rgb:default] Error 1
```

If I uncomment the line it compiles fine.

dexter93, 2023-07-30:
per the chip docs, LPC11uxx doesn't have PAL_MODE_OUTPUT_PUSHPULL support ( at least, specified in the docs), hence the undef there. However, you are able to compile a working(-ish?) qmk build.
[image]

Digging a bit deeper,

commenting out the undef inherits the default [value](https://chibios.sourceforge.net/docs3/hal/group___p_a_l.html#gaedde8679d38abe282c7a70d75a04cda6)
```c
 /**
   * @brief   Push-pull output pad.
   */
  #define PAL_MODE_OUTPUT_PUSHPULL        6U
```
which through qmk's abstraction
```c
#define setPinOutputPushPull(pin) palSetLineMode((pin), PAL_MODE_OUTPUT_PUSHPULL)
```
goes all the way down to the LPC GPIO LLD, forwarding the value to

```c
void _pal_lld_setpadmode(ioportid_t port, iopadid_t pad, iomode_t mode) {
```
The following condition gets calculated as false

```c
  if (mode & MODE_DIR_MASK) {
    LPC_GPIO->DIR[LPC_IOPORT_NUM(port)] |= (uint32_t) 1U << pad;
  } else {
    LPC_GPIO->DIR[LPC_IOPORT_NUM(port)] &= ~(((uint32_t)1U) << pad);
  }
```

Which in turn sets the pin direction to input
[image]

The User Manual for the chip family has some directions on selecting GPIO output

[image]

while not specified by name, it might be possible to enable pushpull output by doing

 ```c
#define PAL_MODE_OUTPUT_PUSHPULL   (MODE_DIR_OUT | MODE_AD_DIGITAL)
```
no clue if it will actually work however.

Re-writing the matrix scan routines in QMK to be open-drain compatible might be the more appropriate approach for your use-case

Rayyanbaig1243, 2023-08-03:
I updated this file

closed 2025-08-09 by crashniels